meshtastic/Meshtastic-Android #777: [Feature Request] app connection directly to mqtt server

Marlontebaldi, 2023-11-10:
It would be interesting if there was the possibility of connecting the app directly to the mqtt server without the need for a radio, and the Android or iOS itself could be a node on the map, to have remote access to the boards when out of range

pavelw, 2023-12-02:
currently Android app has only minimal utilization of the Android device - it gets its GPS location (app option "provide phone location to mesh").

but this is a precedent :)

connectivity to MQTT server via app would be useful in other scenarios.

when app connects to lora device via bluetooth, most lora devices with ESP cannot to have wifi enabled at the same time.

so usual solution is different lora device to have wifi enabled, bluetooth disabled.
this device needs to be always connected to the SAME wifi as this is a static configuration.

**when the android/iphone app would handle the MQTT connectivity it would be much more elastic:**
- anytime android device has access to internet it  could act as MQTT reporting device, or as bridge for the whole mesh
- access to device via app using bluetooth would be possible at the same time

(of course the lora device would need the role of ROUTER or ROUTER_CLIENT in this scenario)

josh208, 2024-01-26:
Looking for the same thing. Use case I was contemplating was that I'm building a Mesh in a remote area with no cell coverage. A lot of people recreate in this area (Snowmobiling / ATVs / Hiking / Fishing / Hunting / etc). It would be great to be able to easily communicate with loved ones that have internet acccess without requiring them to also have radios, or even be in range of the mesh. These would obviously not be repeating anything, but could pub/sub to the MQTT server to be able to send/receive messages to/from the mesh.

sacarlson, 2024-02-15:
I think this would be a very useful feature and should not require that much work to implement.  As I have an app already running on my android phone that works as a mqtt client, so we see that it can be done as seen in MyMQTT and a few other android apps.  I also note the supported mqtt libs in java that will do most of the work as seen here: https://github.com/eclipse/paho.mqtt.android.  This is a bit over my pay grade so I only hope someone or a group takes a bite out of this one.  Good luck to those that give it a try or at least start a branch to work from if not already.

andrekir, 2024-06-08:
Duplicate of #361

closed 2024-06-08 by andrekir