meshtastic/Meshtastic-Android #813: Send Position - not just Request Position

Ferretdog, 2024-01-12:
Long-press on a node allows me to request a position update, but there is no way for me to send my position manually. The iphone app allows the opposite I believe - it allows us to send but not request. 

I would like to be able to manually send my location from the app.

Sending to the default channel is ... ok ... but being able to send it as a message direct to a contact, or to a secondary would be 1000x better.  Perhaps it could be a button inside the chat with that channel/contact, almost like a type of quick chat? Or there could be a an option when generating a quick chat to have that text include your current location (a toggle when making the quick chat).

garthvh, 2024-01-12:
They both do the same thing, send your position and request a response.

garthvh, 2024-01-13:
The other position updates are sent without requesting a response

todd2982, 2025-03-09:
> What if I want to update my own location without updating that of another node?

When you use the "Request Position" function, the radio sends your position to the other radio asking for an acknowledgement.  That acknowledgement from the other node is how you get their position back.  What you are asking for already exists, but I think the documentation is a little lacking in this area.

jamesarich, 2025-03-19:
this functionality exists - however some better labeling of the action might alleviate the confusion
from `Request Position` to `Exchange Positions`

closed 2025-03-19 by jamesarich