meshtastic/Meshtastic-Android #814: Location Ambiguity

Ferretdog, 2024-01-12:
Coming from using APRS in the ham radio world, I often enable Position Ambiguity which rounds off the last x digits of my lat/long. This gives people a rough idea where I am, but not an exact location, which is sufficient for many purposes. I might be happy to have people know roughly where I live without knowing the exact house, or knowing what town I'm in, but not much more than that. 

- I suggest a position ambiguity be added to the location settings, with a user selectable choice of digits of the decimal lat\long from 0 to 5 (or more), eg. x., x.y, x.yy, x.yyy, etc.

For example a decimal lat/long to 4 or more decimal places locates a specific house. [-37.8929, 145.1524](https://www.google.com/maps/place/37%C2%B053'34.4%22S+145%C2%B009'08.6%22E/@-37.8928979,145.1511125,18z/data=!4m4!3m3!8m2!3d-37.8929!4d145.1524?entry=ttu) 

But if I drop it back to [3 decimal places](https://www.google.com/maps/place/37%C2%B053'31.2%22S+145%C2%B009'07.2%22E/@-37.8919979,145.1507125,18z/data=!3m1!4b1!4m4!3m3!8m2!3d-37.892!4d145.152?entry=ttu) I'm still pretty close without disclosing which house I'm in, [2 places is quite adequate](https://www.google.com/maps/place/37%C2%B053'24.0%22S+145%C2%B009'00.0%22E/@-37.8899957,145.1474251,17z/data=!3m1!4b1!4m4!3m3!8m2!3d-37.89!4d145.15?entry=ttu) or even [1 place](https://www.google.com/maps/place/37%C2%B048'00.0%22S+145%C2%B006'00.0%22E/@-37.7940619,145.0664402,13.23z/data=!4m4!3m3!8m2!3d-37.8!4d145.1?entry=ttu) is plenty for what part of town or what suburb you're in which is often enough on a shared channel.

crypto-smoke, 2024-01-13:
I like this, but it's also probably best changed in the firmware repo and then implemented in the various UIs

garthvh, 2024-01-13:
Are you not able to reduce location accuracy in android settings? This is an option for any app using location on iOS.