meshtastic/Meshtastic-Android #819: Feature Request - stop display from sleeping in node/contacts page

b8b8, 2024-01-21:
I know earlier there was a change to keep the screen ON when on the map page. This is really useful. It would also be useful if this functionality was added to the contacts/node page as well. This would make it easier when driving to glance to see if new nodes are discovered or last heard as they are automatically moved to the top of the list. This may make more sense as a toggle as some users may prefer the screen to time-out when on everything but the map page.

andrekir, 2024-09-01:
I see why you'd want the screen to stay on for the node page, but wake locks should be reserved for special cases like for navigation, media playback, etc... where it’s really necessary. we'll stick with the current setup for now.

closed 2024-09-01 by andrekir

jamesarich, 2024-09-01:
You can always adjust your Android system display screen timeout settings as well.