meshtastic/Meshtastic-Android #826: [Feature Request] - More info on node page

b8b8, 2024-01-28:
I posted this on discord and it seemed to have positive feedback. My suggestion is to add a single extra line to the node page for each individual node so the following is shown.

BAT% BATVoltage
Last Heard
LAT LON ALT #SATS
RSSI SNR Channel (and/or freq.)
TEMP HUM PRESS AIR-RESISTANCE (future sensors)

HikerScott, 2024-02-15:
BATVoltage is key... the current structure of simply showing that it is plugged in is nearly useless.  In the field I need to know once the battery is fully recharged and currently the only way to know is to disconnect, reconnect then look at the debug messages.  Quite frustrating.

(Also: Easily seeing last known Lat/Long/Alt is great for SAR)

b8b8, 2024-02-15:
mine shows the battery voltage and percent when unplugged. Is yours not and thats the reason to go to the debug?

HikerScott, 2024-02-15:
Negative.  The problem is it doesn't show the voltage of the battery while charging.  The voltage is available in the debug while charging, it's just not available in the GUI.

We want to keep the devices connected to the charger for as little time as necessary because, as you know, once it gets full it stops taking a charge altogether (and starts drawing down the battery again).  By knowing how close you are to a full charge you know how long you can walk away for (or must stay up at night waiting for it to finish charging).

b8b8, 2024-02-15:
i agree with everything you are saying except the drawing down the battery after. If you keep the nodes plugged in the charging circuit will keep the batteries topped up.

daviscodesbugs, 2024-02-25:
This is interesting... my nodes that are plugged in just show battery at 100% and the corresponding voltage. I always thought that the nodes showing the plug were "hard wired" nodes, not battery nodes that were charging

andrekir, 2024-02-25:
all info come from the radios. batt% value of 101 shows the power plug.

HikerScott, 2024-02-26:
Bingo... And it shows the plug *and no voltage*.

It should always show the voltage so you know how close it is to being full.

________________________________
Canada's Largest and Most Reliable 5G Network

This communication is confidential. We only send and receive email on the basis of the terms set out at https://www.rogers.com/emailnotice<https://www.rogers.com/web/Rogers.portal?_nfpb=true&_pageLabel=NEW_GCT&_nfls=true&setLanguage=en&template=emailnotice>

Le r?seau 5G le plus ?tendu et le plus fiable au pays

Ce message est confidentiel. Notre transmission et r?ception de courriels se fait strictement suivant les modalit?s ?nonc?es dans l'avis publi? ? www.rogers.com/aviscourriel <https://www.rogers.com/aviscourriel>
________________________________

b8b8, 2024-02-26:
what about the other changes to give more info on the node list?

daviscodesbugs, 2024-02-26:
@andrekir Could you reopen this one? That PR didn't include all the changes here

daviscodesbugs, 2024-02-28:
@b8b8 How important is it to actually see the coordinate values? I would love your opinion on this: https://github.com/meshtastic/Meshtastic-Android/issues/545#issuecomment-1969312902

b8b8, 2024-02-29:
i think its really important to show the actual location coordinates. its good for screen shots and provides the raw data. I dont think this info should be dumbed down.

prokrypt, 2024-03-01:
Battery voltage now shows 1 significant digit instead of 3.
3 significant digits is a bit overkill, but showing just 1 significant digit is nowhere as useful as 2...

daviscodesbugs, 2024-03-01:
Oh, yup, you're right. Looks like it used to show 2 and I changed it to 1.
https://github.com/meshtastic/Meshtastic-Android/commit/95c0d182806ea8f2ed16beef3d81819d78ccfbac
I'll set it back to 2

daviscodesbugs, 2024-03-19:
@andrekir 
Not sure why merging #895 didn't close this one, but I believe this is complete

closed 2024-03-19 by andrekir

b8b8, 2024-03-23:
nice additions

PR #874 (2024-02-26): Improved battery/voltage info in node list
Doesn't completely resolve #826, but addresses the voltage info complaint from @HikerScott 

Primarily, moves the battery info into a composable and uses that in the adapter
I added in a couple more battery icons to get a quick idea of the level. I set the ranges arbitrarily, so we could tune these later to match user expectation better
I also cleaned up the adapter item layout for nodes to avoid manually declaring margins on everything. Additionally, moved the sample text to a tools attribute so it still shows up in the preview, without actually sett [... cut]