meshtastic/Meshtastic-Android #829: Question on permissions

IzzySoft, 2024-01-31:
My APK scanner just got some extra checks implemented recently, and on today's update reported:

    ! repo/com.geeksville.mesh_30219.apk declares sensitive permission(s): android.permission.BLUETOOTH_SCAN android.permission.ACCESS_FINE_LOCATION android.permission.ACCESS_COARSE_LOCATION android.permission.ACCESS_BACKGROUND_LOCATION android.permission.CAMERA

Could you please clarify what those permissions are needed for? I mean, some can be guessed if one's a bit familiar what the app does, but not everyone is :wink: 

Thanks in advance!

andrekir, 2024-02-01:
Hello @IzzySoft !

[image]

Meshtastic uses [inexpensive LoRa radios](https://meshtastic.org/docs/hardware/devices/) for long range off-grid communications.

- The app connects to radios via Bluetooth, WiFi or Serial.
- Radios without GPS can use Android's GPS.
- QR Codes containing configs and channel encryption can be shared using the camera.

**All runtime permissions are optional.**

Short video explaining location permissions:

[image]

IzzySoft, 2024-02-01:
Thanks, added to the allow-list then:

```yaml
      android.permission.BLUETOOTH: needed to connect to radios via Bluetooth
      android.permission.BLUETOOTH_ADMIN: needed to automatically pair detected radios
      android.permission.BLUETOOTH_CONNECT: needed to automatically connect to radios via Bluetooth
      android.permission.BLUETOOTH_SCAN: needed to find radios to connect to via Bluetooth
      android.permission.ACCESS_FINE_LOCATION: needed so radios without GPS can use Android’s GPS
      android.permission.ACCESS_COARSE_LOCATION: needed so radios without GPS can use Android’s location service
      android.permission.ACCESS_BACKGROUND_LOCATION: needed to provide radios without GPS with Android’s location while the app is in background
      android.permission.CAMERA: needed by the integrated QR Code scanner
```

> All runtime permissions are optional

… since Android 6. Your app still supports Android 5 :wink: And even being optional: it's good to know in advance, builds up trust :smile: So if you want to cross-check (and maybe suggest some adjustments):

[image]

Oh yeah: `DEPENDENCY_INFO_BLOCK`. A BLOB only readable by Google (as it's encrypted with their public key) and thus useless for everybody else. Want to get rid of that?

```gradle
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

For some background: that BLOB is supposed to be just a binary representation of your app's dependency tree. But as it's encrypted with a public key belonging to Google, only Google can read it – and nobody else can even verify what it really contains.

andrekir, 2024-02-01:
``` yaml
      android.permission.BLUETOOTH_ADMIN: needed to find radios to connect to via Bluetooth
      android.permission.BLUETOOTH_CONNECT: needed to connect to radios via Bluetooth
```

a couple suggestions above. appreciate your interest and support. if there's anything I can help, let me know.

IzzySoft, 2024-02-02:
Thanks, adjusted!

closed 2024-02-02 by IzzySoft

IzzySoft, 2025-05-25:
Our scanners reported:

    ! repo/com.geeksville.mesh_30603.apk declares flag(s): usesCleartextTraffic

I couldn't find a match in the repo here; can you please clarify what non-secured `http` connections are being used and why? Thanks in advance!

jamesarich, 2025-05-26:
Https app links 'https://meshtastic.org/e/' was already being handled for channel import. This just adds another path for contact import handling. Http (clear) was also added to provide more flexibility in case the other clients supply the channel or import URL that way.

https://developer.android.com/training/app-links

It's easy enough to remove the HTTP, and shouldn't impact functionality if it truly is an issue.

IzzySoft, 2025-05-26:
> if it truly is an issue.

That is for you to decide. It's definitely better to have all traffic secured. So plaintext should only be used if it cannot be avoided. An example for the latter are self-hosted resources in a local intranet, where it would be hard to impossible to get proper certificates for.  In such cases, we just need to make it transparent, which we then would take care for if you let us know.

IzzySoft, 2025-05-26:
Playing the devil's advocate: import from where? This could be read as the app importing via http from "some insecure server".

jamesarich, 2025-05-27:
@IzzySoft after consulting with the iOS and web client leads - all of the urls that they generate for use as channel or contact import deeplinks are https, so this is not needed. Will be removed in #1942 .

prokrypt, 2025-05-27:
> Playing the devil's advocate: import from where? This could be read as the app importing via http from "some insecure server".

It imports data from within the url string, which doesn't actually lead to a valid internet resource.
`http://meshtastic.org/e/#CgkSAQEoAToCCCAKKxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx`