meshtastic/Meshtastic-Android #846: Deeplinking is broken

adriantache, 2024-02-11:
Deeplinks, for example for joining a channel, are broken. I have checked, and [the file](https://meshtastic.org/.well-known/assetlinks.json) does exist, but maybe the app signature has changed? If it's not that, and it's the new Android 14 explicit intents, maybe I can try to fix them myself and raise a PR.

andrekir, 2024-02-11:
can you confirm your version and install source (Play Store, APK, F-Droid)?

adriantache, 2024-02-11:
Installed from the Play Store, although it seems I was missing the latest update (and it keeps failing to install, for some reason, maybe my version was signed with a different key?).

andrekir, 2024-02-11:
odd. the release build runs via GitHub actions. no changes there.

andrekir, 2024-02-11:
updating from 2.2.19 to 2.2.20 seems to be the problem. I suspect this is the cause: https://github.com/meshtastic/Meshtastic-Android/commit/d75188f03c80614e33762212fa4df3b10eddc6a6

I'll revert that for the next release.

commit a88ffbc0 (2024-02-12): refactor: disable dependency metadata for `fdroid` flavor only

fixes #846
closed 2024-02-12 by andrekir