meshtastic/Meshtastic-Android #851: Add DNS support for node connection

mattdrum, 2024-02-13:
I would like to have the ability to connect to my gateway node via a DNS name. Currently the UI only allows for an IP address to connect to a network connected node.

mattdrum, 2024-02-13:
My ultimate goal would be to have this node port forwarded it to the internet and use my public DNS. That way as long as I have an internet connection I would be able to have access. The app would need to have the ability to specify the port number as well.

kquinsland, 2024-12-31:
I am a +1 for this as well. For reasons that are not important here, I don't always have long-lived DHCP allocations for nodes on LAN. Ideally, there would be a way to say ["on `$homeWiFi` connect to `$homeMeshtasticNodeHostName`"](https://github.com/meshtastic/Meshtastic-Android/issues/1410#issuecomment-2566747087) where `$homeMeshtasticNodeHostName` is something that the local DHCP/DNS server can map back to whatever IP is correct at the time.

closed 2025-05-24 by github-actions[bot]