meshtastic/Meshtastic-Android #854: [Feature Request]: Select node and send direct messsage from channel window

scotty123, 2024-02-15:
I would find it usefull if you can select a node (message ) in the channel window and then send a direct messsage to the selected node.

Why?  You can only send a direct message starting from the node list and if you use a mqtt channel then the node list becomes very crowdy and takes time to scroll through the list and find the node.

scotty123, 2024-02-15:
#683 is also good but I mean a different use case.

You read a message in the channel and you want to reply via private message. If you could click on the node and select "direct message" then you don't need go to the Nodelist and find the node there.

andrekir, 2024-02-19:
this was added in 2.2.22 with #844

closed 2024-02-19 by andrekir