meshtastic/Meshtastic-Android #859: [Feature Request] Separate channel for Logging + Tracking Traceroutes

ayysasha, 2024-02-17:
I am proposing a UI change surrounding Traceroutes. Traceroutes are an extremely helpful method of quickly determining traffic flow in a given mesh and troubleshooting connections. Currently in the app, the traceroutes are performed by long pressing a node, selecting traceroute and then waiting for a response. If the Traceroute is successful, the user is presented with a dialouge popup with the informaiton. If it fails, they are not presented with anything, sometimes creating a condition where a user is waiting for a response for an indetermined amount of time.

My proposal is as follows: A separate "Channel" for traceroutes that logs the information regarding them, and also indicates when a traceroute is sent and no response is received. @avianomy created these mockups below:

alt txt:  A screenshot of the messages tab of the meshtastic app, with an additional "channel" labeled "traceroutes".

[image]

alt txt: A screenshot of a chat interface that includes traceroutes.

[image]

Key features:

- Centralized location to look at traceroutes and their history.
- Indication if a user initiated traceroute has failed or not.

ayysasha, 2024-02-20:
If the response is not received, that should be indicated to the user as a failure to receive a response for an action they performed, like in the messages UI.

The logs in the debug channel VS it being somewhere else (as mentioned in the FR, could be a separate tab) does not matter.

The android app has context into:

- The traceroute going out
- The traceroute coming back in, if ever

The debug panel does not have a search function/way to filter types of messages. Having a separate tab for these will be a serious QOL increase over the current workflow of waiting for a response that may never come on a UI panel you just have to stare at for x time until you get bored/receive a response.

ayysasha, 2024-02-20:
Also - This is already a feature on IOS under "Trace Route Log"

GUVWAF, 2024-02-20:
> If the response is not received, that should be indicated to the user as a failure to receive a response for an action they performed like in the messages UI.

It's not so simple. For messages on a channel, the firmware computes when an implicit acknowledgment from a direct neighbor (cloud with checkmark) should have been received based on the airtime of the packet. It then tries to retransmit up until three times, after which the packet is marked as failed. For direct messages, you may also get the real acknowledgment (person with checkmark), but it will not be marked as failed when that doesn't arrive, simply because we don't know when it arrives, as it may go over multiple hops. The same holds for a traceroute. In the worst case it may take 16x times the airtime of the packet (7 hops to and back), plus additional processing or delays due to other ongoing transmissions, etc. Sure, the firmware could inform the app that it likely failed after that time, but that will probably be long after the user already tried it another time.

ayysasha, 2024-02-20:
Thanks for your response.

_- For messages on a channel, the firmware computes when an implicit acknowledgment from a direct neighbor (cloud with checkmark) should have been received based on the airtime of the packet. It then tries to retransmit up until three times, after which the packet is marked as failed. 
-  The same holds for a traceroute. In the worst case it may take 16x times the airtime of the packet (7 hops to and back), plus additional processing or delays due to other ongoing transmissions, etc._

Yes, and currently the user must stare for 16x the airtime of the packet on a panel that does not indicate to them if there has been a response. I understand the limitations of the firmware but this is agnostic of firmware capability. There should be a place a user could "click out" of the app and go back to in order to see both Traceroute history and if a requested traceroute has received no response after x time. I understand the firmware will not report the failure, but the android application has context into **if** a traceroute has been sent and **when** a traceroute has been sent.

_- Sure, the firmware could inform the app that it likely failed after that time, but that will probably be long after the user already tried it another time._
I hope we can both agree here this is not great UX - the user should not have to re-request traceroutes and stare at a panel for 16x of the airtime until they get tired of it to determine if there is a failure or not. a user should be able to request a traceroute and, **like a message, get notified when they receive a response** and have that presented to them in a digestible way (again does not have not be in the messages tab, maybe its own dedicated tab or submenu).

Real user stories regarding field use of the android traceroute functionality have been posted in discord. I personally experienced this when trying to manage a fairly large network in a field environment.

GUVWAF, 2024-02-20:
> There should be a place a user could "click out" of the app and go back to in order to see both Traceroute history and if a requested traceroute has received no response after x time. 

I see, I can’t speak for the feasibility of this in the Android app. 
I just wanted to clarify why it can’t reliably mark a traceroute as **failed**, it can merely show “No response” until it got one, like the iOS app does.

ayysasha, 2024-02-21:
Sure, glad we got that clarified.

In the case that the rest of my proposal is agreeable, can we un-discard this FR? I think that this is good functionality and the IOS app already has 80% of this feature. @andrekir

ayysasha, 2024-02-26:
Any update on the above? Can we reopen this now that the FR is clarified?

ayysasha, 2024-03-05:
@andrekir do you find the above clarification with @GUVWAF acceptable as a feature request?