meshtastic/Meshtastic-Android #860: [Feature Request]: Graphical Signal Strength/Quality

KevinKW87, 2024-02-17:
I noticed the Apple app has an easy to read graphic for understanding signal strength. After asking garth about it, here's the information he gave me:

[](https://sensing-labs.com/f-a-q/a-good-radio-level/)

The per preset snr numbers are from here [](https://www.thethingsnetwork.org/docs/lorawan/rssi-and-snr/)

Here are the preset values [](https://github.com/meshtastic/Meshtastic-Apple/blob/2b8118b3f45359bb835bac9c18784ec8e67be724/Meshtastic/Enums/LoraConfigEnums.swift#L214)

All the color and text calculations are here
https://github.com/meshtastic/Meshtastic-Apple/blob/2b8118b3f45359bb835bac9c18784ec8e67be724/Meshtastic/Views/Helpers/LoRaSignalStrengthIndicator.swift#L42

[image]

[image]

asjmcguire, 2024-03-04:
Just chiming in here. While the signal level looks pretty on iOS, it also significantly draws attention to the signal level... which leads to much more confusion. So the thing that most people don't get, and I'm not sure if it is really documented anywhere - the signal level is a lie, and causes frustration.

Example:
Node 1 (You) can see Node 2 (RSSI -50)
Node 1(You) can see Node 5 (RSSI -50)
You are told, that the signal between you and Node 5 is excellent, but that's a lie, your signal to Node 2 is excellent, but you have no idea what the signal is like between Node 2 and Node 5, Or Node 2 to Node 3, Node 3 to Node 5.

The RSSI and SNR you see is between your node, and the LAST node to relay a packet to your node, in most cases, it is NOT the RSSI and SNR between you and the node you are trying to reach. This leads to people getting very frustrated because they think they have an excellent signal to a node but can't send any messages to it.

There is firmware in the works that will actually tell you the hop count for every packet, so instead of relying on an RSSI and SNR figure which is completely unhelpful unless the node really is only 0 hops away, we will instead get told the much more useful information of how many hops away every node is.

Bjarne-Winkler, 2024-03-29:
I am with @asjmcguire regarding the frustration of what is the real rssi/snr in the route the signal has taken to get to my Node.
The first fix we need in the Android App should be to show signal strength like the iPhone and it can be as simple as red/yellow/green font.  It would be nice if the user of the Android App could set (as in App configuration) the levels, but that can be down the road.
The second fix will most likely take Firmware changes, and that is, we need Hop information. If I have a Router|Client on the rooftop, then just getting the rssi/snr between my indoor unit and the Router|Client at the rooftop means nothing!  We need to see each hop rssi/srn information.  Let's say that the protocol can't hold all the information, then maybe we can spare one bit for Good rssi/snr or Bad rssi/snr.

asjmcguire, 2024-03-29:
There is no spare space in the header for any more information now. the HopsAway feature was added to the 2.3.X firmware and 2.3.2 was promoted to stable (beta) today. So now we just have to wait for people to start updating. HopsAway will still be more useful information than what we have right now (and will hopefully mean people will stop doing traceroutes just to see how many nodes a signal is going through).

closed 2024-11-21 by andrekir