meshtastic/Meshtastic-Android #861: when connecting new node make "provide phone location to mesh" default off

costonisp, 2024-02-18:
When  the box  "provide phone location to mesh"  is ticked, this setting in the app is remembered when connecting to another device. 
So the location in a newly connected node is also updated  to the phone's location, unless you remove the checked box before disconnecting your old device. 
I suggest to make the status of this setting by default to off when connecting to a new device. So only when a user knowingly want to set the location with a phone, the box must be ticked. This prevents that the device location is set/changed unwillingly when the user forgot to remove the tick before.

geeksville, 2024-02-18:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/bug-detected-in-position-used-fixed-position/9417/33

Kookcmoi, 2024-02-18:
For information the page on meshtastic discourse where this matter was explained has been censored by "Garth", the local moderator who could not stomach the bug exposure. He made the post invisible but it is still accessible here, for memo and archive : 
https://meshtastic.discourse.group/t/bug-detected-in-position-used-fixed-position/9417/32

erayd, 2024-02-19:
For what it's worth, I also consider the current behavior to be a bug. "Fixed" should mean precisely that.

@Kookcmoi I saw that. Could understand (albeit disagree with) locking the thread if they didn't want to have that discussion, but hiding it seems fairly unreasonable. The problem won't vanish just because they choose to ignore it; all hiding it will do is ensure the debate flares up elsewhere.

haplm, 2024-04-01:
There is another fairly valid usecase for FIXED being really fixed. I have a node that is connected via http (ie. BT disabled), and it doesn't have a GPS. Yet I would like the position of the node to be reported. At the moment, I'm out of luck, because I can't set lat/long via the web interface after the node reboots.

It would be great if fixed position would be really fixed, it would disregard status of GPS, disregard what potentially connected app is sending, and lat, long and alt would be persisted in the configuration (which is not the case at the moment)

TonyHolden132, 2024-04-21:
The app works as advertised, provides location of a node to the mesh if a box is checked.

If you don't want it to do that, uncheck the box. 

Everything else is semantics.

costonisp, 2024-04-22:
> The app works as advertised, provides location of a node to the mesh if a box is checked.
> 
> If you don't want it to do that, uncheck the box.
> 
> Everything else is semantics.

I do not doubt that the app works as intended/advertized.
But if you manage several nodes, with fixed location, with a single phone (which may be not as intended by the developers) it can be very annoying to find that the node you connect with, takes the location of your phone which may not be as intended by the user.
So it would be much more intuitive if the app switches off the "provide phone location to mesh" box automatically when connecting to a (new/different) node.

garthvh, 2024-04-28:
Now that fixed position is not over written by a phone location this value survives switching nodes on both iOS and Android.

closed 2024-04-28 by garthvh

Firecul, 2024-04-28:
> Now that fixed position is not over written by a phone location this value survives switching nodes on both iOS and Android.

A lot of nodes I see near me don't have a location set as far as I know. What would happen in that scenario?

rjDipcord, 2024-07-31:
> The app works as advertised, provides location of a node to the mesh if a box is checked.
> 
> If you don't want it to do that, uncheck the box.
> 
> Everything else is semantics.

Except if you're not already connected to a node. You cannot 'uncheck' the box unless you're already connected. So if someone doesn't want a node to have it's location updated, they can't turn it off. "Share phone location to mesh" should be persistent on a per-node basis. Not global to the app.