meshtastic/Meshtastic-Android #862: Cannot enter fixed position unless device already has a position set

erayd, 2024-02-19:
If the device does not already have coordinates saved, then the coordinate fields for "fixed position" location are disabled and cannot be edited. This means that it is impossible to remotely provision the location on a non-GPS device, unless that device has first obtained its location from a paired phone etc.

App version: 2.2.23
Firmware version: 2.2.21.7f7c5cb
Device: Heltec V3

[image]

andrekir, 2024-02-19:
it is not possible to set the position for remote nodes

erayd, 2024-02-19:
@andrekir Thanks for the speedy response 🙂. Out of curiosity, why not?

andrekir, 2024-02-19:
to set the coordinates you send a position packet from yourself to yourself. firmware allows that over the API (Bluetooth, TCP/IP, Serial) but not over LoRa, otherwise nodes would be vulnerable to anyone changing their position (and time).

haplm, 2024-04-01:
This unfortunately doesn't work over serial or TCP/IP either (from a computer). The node resets, but nothing happens. I'm using

`meshtastic --setlat 51.0 --setlon 14.4`

basementbuilder17, 2024-04-01:
> This unfortunately doesn't work over serial or TCP/IP either (from a computer). The node resets, but nothing happens. I'm using
> 
> `meshtastic --setlat 51.0 --setlon 14.4`
> 
> And of course fixed position is true as well.

i found that if you are connected to the device using BLE at the same time as Serial (USB), the location will not update. Was very confusing/frustrating because it responded as if it was working in python. Eventually I deactivated the bluetooth from my phone and the commands worked. Also found you needed to disable fixed position, set the position, then enable it again. hope this helps someone else!

haplm, 2024-04-01:
Yes, I have also found out that disabling the fixed position is an absolute must. Without it, it won't report position even if pushed by phone. IMO the whole "Fixed position" thing is completely broken and should be reworked.

Few more notes:
- Contrary to iOS, macOS and web, Android client can connect and push position also over IP!
- Android generally has much more advanced options compared to iOS/macOS/web
- As long as Fixed Position is enabled and lat/long is set, Android client shows the set position, but it doesn't get reported to the map (or to other nodes)
- Fixed position needs to be disabled to get position reported. I'm not sure what re-enabling it would do

basementbuilder17, 2024-04-01:
> Yes, I have also found out that disabling the fixed position is an absolute must. Without it, it won't report position even if pushed by phone. IMO the whole "Fixed position" thing is completely [...]

agreed i am new to this and the hardest part of setup was properly setting a correct, fixed location. definitely felt buggy. 
unfortunately no access to android, but i was able to get a good procedure going with the iOS app + CLI.

VigibotDev, 2024-06-03:
+1 here, I have remote admin node and I need to set the position to remote get neighbor info on meshmap.net

callimero, 2025-08-27:
I also set up a device that has no GPS. I used a rough position from the map and now I am stucked there but would need an exact position. It seems I can't push my phone's position via meshtastic app to the device either which is a shame.

garthvh, 2025-08-27:
> I also set up a device that has no GPS. I used a rough position from the map and now I am stucked there but would need an exact position. It seems I can't push my phone's position via meshtastic app to the device either which is a shame.

Fixed position fixes the position, to update the position you need to disable fixed position.