meshtastic/Meshtastic-Android #867: [Feature request]: Add "Delete" to long press menu while in node list.

BillyBag2, 2024-02-22:
# Long press menu in node list has an option to delete node option.

The iphone app has the ability to delete a node off the radio. After deleting the node it does not appear in the node list on the app or on the list acquired over serial by the CLI. The node will be appear again if new information about the node is received.

Looking at the code it seems to send an admin message with "remove_by_nodenum" or something like that. (Not sure how it all works.)

Bjarne-Winkler, 2024-03-28:
Android App: Remove Node or even Block Node from the Network List.
Both options would beat the NodeDB Reset function that wipes everything.
Since we know that Meshtastic Cli can delete single nodes in the device Network List, then the firmware would support the removal of a Node from the Network list.  Blocking has to be the App's job.

andrekir, 2024-04-28:
added in #958

closed 2024-04-28 by andrekir