meshtastic/Meshtastic-Android #890: Feature Request: Display paxcounter data on node list.

Donnie650, 2024-03-01:
Display Paxcounter data like telemetry in the node list. Make it a on/off switch in paxcounter module. this will allow it to remain out or in the node display on remote units. Thanks.

Donnie650, 2024-04-10:
What is the issue with adding this so netwok can see? its been active from 2.1.16 we are now at 2.3.5 and I still can't see it on the android app. please update so the pax counter has some usefulness on meshtastic android app and on the mesh. Running android app 2.3.4

closed 2024-06-11 by Donnie650

tmedicci, 2024-12-27:
Why was this issue closed? As far as I know, it isn't available yet, right?

tmedicci, 2024-12-27:
> Why was this issue closed? As far as I know, it isn't available yet, right? (@Donnie650)

MagooYFZ, 2025-01-07:
whats up with this ?
could we get the paxcounter information on the android app ?
Or some documentation on how we can use this data.
Thank You
@Donnie650

Donnie650, 2025-01-07:
I closed it because I gave up tying to get this on the app. I keep being told its working. I have yet to see that on any android app version. Disappointment all around. I'm not one of the programmers just a user trying to get this working.