meshtastic/Meshtastic-Android #906: Export/Import Configs not restoring everything

Danie10, 2024-03-10:
Just saved configs on Android v2.2.23 with everything selected. After flashing firmware v2.2.24 with full wipe, I imported config again, but some settings were not restoring e.g.:

1. Fixed location reset to 0 and 0 with 0 altitude.
2. Bluetooth "name" did not restore (but it did restore my BT fixed PIN).
3. Store & Forward did not restore as enabled, and no of records etc was reset to 0.
4. Neighbor info config also did not restore and was disabled.

talkingnews, 2024-03-11:
Looks similar to the issue I opened last week in the CLI section: https://github.com/meshtastic/python/issues/493

Danie10, 2024-03-11:
@talkingnews yes definitely similar I'd say. I had listed shortname and Bluetooth name as an issue under Android, but removed that after observing it came right a few minutes later after having imported the saved config file. In this case, that was some syncing/reboot that resolved it.

But if the Android app itself is saving and importing configs, it should do all the supported configs in the app. I'm suspecting newer features have been added, and the export/import in the app has not been correspondingly updated. It is likely the same for the CLI side as well. It is probably a change control that should trigger when features are added or changed, to stay in sync (whether Android, iOS or CLI).

Danie10, 2024-03-11:
@andrekir OK thanks:
1. Could be useful having location when it is being used as a fixed location device (fixed location toggle in settings is on), such as at home or other locations. Maybe otherwise with addition of a toggle when exporting, to include or not.
2. I'll check this properly on next upgrade.
3. Thanks

closed 2024-03-12 by andrekir

joneyes, 2024-07-18:
_Fixed location reset to 0 and 0 with 0 altitude.
that is expected, the location details (lat/lon/alt) are not part of the exported [DeviceProfile]_

i would argue that this is a little short sighted. If the values are set, then they should be exportable.
To me, the whole point of being able to export the config, via Android, Apple or CLI is to save a config that can be quickly restored. It gets very painful when you import the config and have to enter a load more by hand.
This is especially painful for lat/long, since sending the location from the phone API is also not persistent.

Danie10, 2024-07-22:
I agree - I just realised yesterday one of my nodes (one with fixed location) had been on 0 0 coordinates for the last month since I did the update. I forgot the coords get cleared. The intent behind "fixed" is that they are fixed, and not reset. It is also not easy to retype two lengthy numbers in every time as I have to type it in on my phone in the app. Really hope that this can be "fixed".

commit 902d1697 (2024-03-12): fix: add new modules to `DeviceProfile` import

closes #906