meshtastic/Meshtastic-Android #907: Text formatting on node screen appears broken on 2.3.0

eldoctorcool, 2024-03-12:
Attached image is how 2.3.0 now looks... vs 2.2.24
2.3--> Broken...
[image]

2.2.24 ---> works.
[image]

tropho23, 2024-03-12:
Try uninstalling the app, then reinstalling it fresh. I am running the same version of this app and do not experience this problem.

If the problem persists please provide your phone make and model, your Android operating system version, and anything else you think might be relevant.

prokrypt, 2024-03-12:
This is due to a combination of small screen and large font size. Setting a smaller system font in Android settings should make the coordinates of other nodes stay on one line. The info on the connected node, however, looks like it needs some layout/wrapping adjustment.

prokrypt, 2024-03-12:
Some examples: largest display, larger (largest) font size
[image]

largest display, large font size:
[image]

eldoctorcool, 2024-03-13:
Reinstalling the app, cleared cache, etc. all that didn't do anything. This is an Android 7.1.2 generic unbranded device that has worked fine with the Meshstastic Android app for over a year now. Device physical screen is 3 1/4 inches. The font size and screen size are just a notch above default size, (not largest) but this problem also happens with default font and screen size. 

Given that 2.2.24 worked just fine but 2.3 doesn't, I would to please ask if someone could please fix this? Thank you very much.

Hardbuckle, 2024-03-13:
Same issue. Using portrait caused text down the screen. All OK until 2.3.0. 
OnePlus Open in front screen portrait (normal use). Tried all of the above. All  OK previously.

Hardbuckle, 2024-03-13:
[image]

eldoctorcool, 2024-03-19:
Has this been addressed in 2.3.1? Thanks.

eldoctorcool, 2024-03-27:
2.3.1 and 2.3.2 tested. The formatting still broken.

Bryang0syr, 2024-03-30:
Last night 2.3.2 was pushed out onto my Samsung A13 so I woke up this morning to find the above formatting issues. I presume 2.3.0 and 2.3.1 had not been pushed onto my phone when they were released as this is first time I've been aware of the issue. Format is displayed ok in Landscape mode so only an issue in portrait mode

Hardbuckle, 2024-03-30:
Agreed. Formatted still broken in 2.3.2 portrait. Very annoying when half the screen is lost to one node.

francisuk1989, 2024-03-30:
I have no issues on my google pixel 8 pro
[image]

asjmcguire, 2024-03-30:
There's a chance it was fixed for us beta testers in https://github.com/meshtastic/Meshtastic-Android/commit/e3f495e35261e46ae818c7d32b96ee9a1bae472a

eldoctorcool, 2024-03-31:
> There's a chance it was fixed for us beta testers in [e3f495e](https://github.com/meshtastic/Meshtastic-Android/commit/e3f495e35261e46ae818c7d32b96ee9a1bae472a)

It is not. I've tried and 2.3.2 has that fix, its not fixed.

francisuk1989, 2024-04-03:
A new beta has been pushed out sometime on the 2nd of April via Google Play Store, How's everyone find it now if on the Beta?

If you not on the android beta, Go to Google Play Store > search for meshtastic- > scroll down to Join Beta Program and then give it 10mins and there should be a update, if you can also leave the beta at anytime.

Hardbuckle, 2024-04-03:
No improvement. The same issue on 2.3.3
[image]

eldoctorcool, 2024-04-03:
Same here. No improvement. 

I get my beta/alpha builds directly from github. So, whatever its latest on the app, I get to test.

andrekir, 2024-09-01:
this was fixed in recent releases. closing.

closed 2024-09-01 by andrekir