meshtastic/Meshtastic-Android #912: [Improvement] Show nodes that have been seen but for which there is no NodeInfo

ianmcorvidae, 2024-03-14:
If I've gotten an ack or other packet from a node but haven't yet gotten a NodeInfo packet, it seems that it won't show at all in the tab. This matches what the CLI does (though I'll be putting in a feature request there too), but does not match the web client (which shows "UNK: <integer node id>") nor iOS (which shows "Meshtastic XXXX"). I think it's valuable to see those nodes in the list so it's less mysterious who's acking things or to identify nodes that you have a poor connection to (hence no NodeInfo) but see sometimes, as those nodes are the exact sort that smaller improvements like mast height or antenna upgrades can help be seen more reliably.

prokrypt, 2024-03-14:
Yes! This can be quite useful especially when position packets are received from "unknown" nodes, as currently those positions packets don't make it onto the map.

legop3, 2024-03-18:
It also would be nice to have the last seen time be updated by any packet from that radio, instead of what I assume are just position reports.

commit de3d74b9 (2024-05-19): refactor: show nodes with unknown user info

closes #912
closed 2024-05-19 by andrekir