meshtastic/Meshtastic-Android #918: GPS position shown as values close to 0,0 when node has no gps position but GPS precision is set to LOW_PRECISION or MEDIUM_PRECISION

Pscope, 2024-03-17:
Tested in Android App Version 2.3.0. 

If a node is configured on the primary channel with either LOW_PRECISION or MEDIUM_PRECISION, this appears to set the GPS POSITON to 0.10486,0.10486  or  0.00328,0.00328 respectively and is shown in the node list and on the map. 

If HIGH_PRECISION is set when no GPS POSITION is given the node position is 0,0 and so is not show in the node list or on the map.

Maybe the app could also ignore 0.00328,0.00328 and 0.10486,0.10486 as thease are not generally valid positions of a node?

[image]

GMart, 2024-03-20:
I believe this was fixed with this change: https://github.com/meshtastic/firmware/pull/3325
Are all of those nodes updates to 2.3.0+?

Pscope, 2024-03-21:
Yes, all on 2.3.0+ now. and after getting a GPS position and then loosing it again ( indoors) the position is no longer showing incorrect values.  Looks to be resolved now Thanks.

closed 2024-03-21 by andrekir