meshtastic/Meshtastic-Android #922: Hops Away

asjmcguire, 2024-03-19:
The Android app currently lacks the display of how many hops away nodes are.
The thing that would would make the most sense is to only display RSSI and SNR when hops away is zero, or missing. Because obviously since when it is 1 or more hops away, then the signal strength no longer makes sense, people think they have an excellent connection to a node, but really they only have an excellent connection to the 1st hop, the connection between the 1st hop and the destination node, or just the next hop, might be terrible, but that information is never communicated (or available) to the sender.

It makes more sense to show the number of hops in place of the RSSI and SNR when the hop is higher than zero.

prokrypt, 2024-03-19:
I think the RSSI and SNR readings in those cases still can be useful--they just need to be somehow linked to the neighbor that repeated the packet.

asjmcguire, 2024-03-19:
> I think the RSSI and SNR readings in those cases still can be useful--they just need to be somehow linked to the neighbor that repeated the packet.

Which if I understand correctly from the firmware channel on discord, is not information that is available. The only node information is the original sender, nothing from any node the packet has passed through. Maybe V3 will change this, but there is nothing in V2.

francisuk1989, 2024-04-02:
So many features we are missing out compared the iOS version
- Emojis
- root topic recommends
- Sending a GPS location

and more

[image]

DayleDrinkwater, 2024-04-09:
After fiddling around for the past few days, I've finally worked out how to get the data. I'll tidy everything up and make UI elements for it. 
[image]

joneyes, 2024-04-10:
Good result.
One of the other devs has also added the info to the python cli --nodes table

closed 2024-04-21 by andrekir