meshtastic/Meshtastic-Android #926: [Improvement] View maps a zoom beyond tile max zoom

danwelch3, 2024-03-20:
Background: When multiple nodes are close by each other their markers on the map overlap and some are not visible at some zoom levels.  I find this most often on the USGS Topo layer which only zooms to 15.  Since this is one of the few downloadable layers, it's not easily remedied.

Details:  When submitting a PR to increase the max zoom of the USGS Topo layer, I was informed that the USGS tile server doesn't work >15. When looking into this I found that their [map server spec](https://basemap.nationalmap.gov/arcgis/rest/services/USGSTopo/MapServer) specifies zoom levels up to 23, but in testing anything with a z of 16 and below seems to work just fine within the US (areas outside the US are max 12. (Side note, if anyone can give me more info on why it's limited to 15 instead of 16, I'd like to know)

Just for reference, here's an example tile at z=16
https://basemap.nationalmap.gov/arcgis/rest/services/USGSTopo/MapServer/tile/16/24567/13229

It looks like osmdroid supports zooming further than the max tile zoom. This can be seen in their demo app under the "Zoom Restricted Online Tiles" sample.  Is there interest in decoupling the max view zoom from the max layer tile zoom?  If so, I will work on a new PR.

andrekir, 2024-03-21:
the maximum zoom level of an osmdroid `MapView` is set using the `maxZoomLevel` property

danwelch3, 2024-03-21:
Exactly.  This can be accomplished by changing line 509 in MapFragment.kt as follows.
<pre><code>
    fun loadOnlineTileSourceBase(): ITileSource {
        val id = mPrefs.getInt(mapStyleId, 0)
        debug("mapStyleId from prefs: $id")
        return CustomTileSource.getTileSource(id).also {
            <s>map.maxZoomLevel = it.maximumZoomLevel.toDouble()</s>
            <b>map.maxZoomLevel = it.maximumZoomLevel.coerceAtLeast(20).toDouble();</b>
            showDownloadButton =
                if (it is OnlineTileSourceBase) it.tileSourcePolicy.acceptsBulkDownload() else false
        }
    }
</code></pre>

This has the added benefit of not changing zoom as drastically when switching between maps.  If this seems like a change worth making, I will submit the PR.

eldoctorcool, 2024-03-25:
This used to be like this in 2.1.x app IIRC....

andrekir, 2024-10-06:
completed in https://github.com/meshtastic/Meshtastic-Android/commit/be6ea79c04e81463716d7fce52f29ea9878793f1

closed 2024-10-06 by andrekir