meshtastic/Meshtastic-Android #927: Distances of nodes don't update when your node moves.

asjmcguire, 2024-03-21:
I've been out "wardriving" today and noticed that the distance of my home nodes stayed at 295m until I closed the app from the multitask screen and re-opened it. I assume because the distance to other nodes is not calculated unless a packet arrives from them, causing a redraw to happen. I assumed that the distance to other nodes would be dynamically updated as my node moves further away or towards those nodes.

(This calculation is happening on the node itself, when I scroll the screens on the Heltec I can see it has distances of 1.3km which is what the Android app showed after the close and open).

eldoctorcool, 2024-03-25:
I've experienced this, or a form of this since I tired 2.2.24 on all the vehicles. Solution was to reduce smart position distance intervals to 15 meters and 10 seconds, and a 3 second GPS acquire interval... after that, it sort of worked like it did on 2.2.23 with 75 meters and 15 seconds interval, with 10 second GPS interval...

andrekir, 2024-06-30:
fixed in 2.3.14

closed 2024-06-30 by andrekir