meshtastic/Meshtastic-Android #930: [Bug] Trying to change fixed position coords fails

prokrypt, 2024-03-24:
Similar problem as https://github.com/meshtastic/firmware/issues/3468 due to https://github.com/meshtastic/firmware/pull/3403

Fixed position must be disabled and then re-enabled with the new coordinates or otherwise they won't stick.

Possible solution (don't know how the android app sets position) is to have it automagically disable fixed position before setting location and re-enabling? If that's not possible then maybe gray out the input boxes when fixed position has already been set.

garthvh, 2024-03-24:
You can't set a fixed position while fixed position is enabled anymore.

jane1e, 2024-03-24:
why?

prokrypt, 2024-03-24:
> why?

Because someone complained about phone-provided location overriding fixed position coordinates.
?!?

closed 2024-04-01 by andrekir

joneyes, 2024-04-14:
> > why?
> 
> Because someone complained about phone-provided location overwriting fixed position coordinates. ?!?

The way around this would be to make provide phone location to mesh check box always visible (rather than only after device selection), that way you know the status before selecting the device

PR #946 (2024-04-01): refactor: add fixed position admin messages
reference: 
- https://github.com/meshtastic/firmware/pull/3403
- https://github.com/meshtastic/firmware/pull/3490

closes #930