meshtastic/Meshtastic-Android #931: [Issue]MQTT Map reporting interval (seconds)

ramdor, 2024-03-24:
I had noticed some increased outbound data from my wifi heltec v3 node, over 400Mb per 24hrs which was not normal.

I tracked it down to the setting 'Map reporting interval (seconds)' in the MQTT settings. I had map reporting set to MED_PRECISION but I had not noticed that **0** had been used for the interval. I changed it to 900, and outbound data rate dropped to normal low levels.

I assume this got assigned to 0 after I did an update to both firmware and android version the other day. Perhaps it needs a more appropriate value for initial install/update? Note: I did not erase the node when doing the update to the FW.

FW heltec-v3 2.3.1.4fa7f5a
Android 2.3.1

GUVWAF, 2024-03-24:
From 2.3.2 on, the firmware now has a fail-safe to fall back to the default if 0 is set by a client.

ramdor, 2024-03-24:
> From 2.3.2 on, the firmware now has a fail-safe to fall back to the default if 0 is set by a client.

will that resolve endless data being sent to MQTT which seems to have been happening in my case?

GUVWAF, 2024-03-24:
Yes, then it will send it only once every 15 minutes.

ramdor, 2024-03-24:
> Yes, then it will send it only once every 15 minutes.

ah great, I re-read your initial reply and realised what you were saying defaulting to a safe value. Good stuff. Cheers @GUVWAF

GUVWAF, 2024-03-26:
Please close the issue if it's resolved with 2.3.2.