meshtastic/Meshtastic-Android #936: Adding a Raspberry Pi Pico using IP address to the Android App is not working

Sparqx73, 2024-03-26:
Hey guys, I wonder if you can help me 🙏 I have setup a pico w with the waveshare Lora board using the client.meshtastic.org interface and that is all good and I have enabled wifi and have the IP address. However I would like to now add this device to my android meshtastic app but when I go into the app, tap on the settings cog, the option for IP address is greyed out...I can type an IP address in but still can't tap on the IP address check box....am I missing something? Thanks in advance.

GUVWAF, 2024-03-27:
I've noticed that after entering it, then clicking away from the field, then removing a digit and re-entering it, you can select it.

Sparqx73, 2024-03-29:
I upgraded my version of Android OS and the version of the Meshtastic app and tried again and it is now working. As soon as I enter the last digit of the IP address the radio button appears and I can now select it.

closed 2024-03-29 by Sparqx73