meshtastic/Meshtastic-Android #940: WiFi not connected

M1CJX, 2024-03-30:
Attempts to connect and gets to module config 1(3/13) and then Not connected. May very very briefly connect but then disconnects.

Connecting via wifi using Andriod 13TP1A.220624.014
Phomne Redmi Note 10 5G Model M2103K19G
MIUI Version 14.0.4
Andriod Security update 2023-09-01
Meshtastic App 2.3.2
Heltec V3 running 2.3.2.63df972 Beta but I have tested other versions.

I can connect using Chrome web browser without a problem but the app does not connect/stay connected to the IP address.

I have rebooted the router, restarted the mobile phone, checked for updates and moved the node closer to the router.

I am still having consistent problems connecting. Has the app changed in the last few days?

M1CJX, 2024-03-31:
I think the problem is caused by having more than one app connecting via WiFi. I had recently installed the Meshtastic app on a tablet and although this also had problems connecting, I then disabled the app on the phone and the tablet starting working correctly. This is after completely rebuilding the Heltec V3 which did not make any difference. Strangely you can connect using a browser and have the app connected at the same time but it's a problem if more than one app is connected. I hope this helps. Tony (M1CJX)

andrekir, 2024-04-01:
firmware supports only one connection at a time. when a new connection is made, the old one gets dropped.

closed 2024-04-01 by andrekir