meshtastic/Meshtastic-Android #943: Error reporting height above sea level in Meshtastic-Android 2.3.2 when deriving the height from the Android phone GPS

Bryang0syr, 2024-04-01:
Since the version 2.3.2 was pushed out last weekand the height is now displayed on the Android App. I have found the height here is overestimated by around 43 metres. I have checked some adjacent nodes around me and they are reporting 43 metres higher than they should. I checked with a stand alone height app on the android phone which reports the correct height. Using 2.2.23 firmware in Heltec V3

andrekir, 2024-04-01:
there are different models for measuring altitude https://en.wikipedia.org/wiki/Height_above_ground_level

Android [Location](https://developer.android.com/reference/kotlin/android/location/Location) `altitude`: _**"The altitude of this location in meters above the WGS84 reference ellipsoid."**_

different models could explain the difference you are seeing, but either way, that's the information coming from the GPS.

closed 2024-04-01 by andrekir

Bryang0syr, 2024-04-04:
Hi there, I have read up a bit about different Datums and it does not seem a good choice to display a simple ASL figure on the android app. In the UK all mapping references a single Ordnance Survey Datum any mapping app or GPS device such as Garmin is sold with this datum built in so a normal user never gets to see other datum references. Trying to use the fixed location to correct the display to local heights has issues, in that if I select 'use fixed position' and edit the Altitude it reverts to the WGS85 figure from the phone. I have to remove the permissions in the phone for Meshtastic to have access to GPS before the corrected figure is displayed. This has a knock on issue when controlling several nodes such as a mobile node in that I have to remember each time I go out to turn on Meshtastic permissions once I've disconnected from the home node. If I return home with GPS position access still granted then my fixed node reverts to the WGS85 height and I have to manually edit the figure again. This does not seem sensible. As no-one knows whether the displayed figure has been locally corrected or not it is meaningless. Nodes located round our coast at sea level are suddenly 120ft above sea level. I appreciate this is a difficult world choice when writing software but please consider removing this display in future versions as it is meaningless and confusing