meshtastic/Meshtastic-Android #956: [Bug]: Some nodes showing incorrect and matching GPS coordinates

djripcord, 2024-04-08:
I have noticed in the android app V2.3.2 that some nodes are reporting incorrect position data. This would normally not be a concern. What if the GPS fix was inaccurate? Or the target was last heard in that location?

In this situation, I do not believe the above to be accurate. I am witnessing my own portable node with GPS on-board. It is a T-Beam Supreme running 2.3.2.63. It normally reports it's location correctly on the device itself. I can see the coordinates listed on the LCD match my real position. However the app is showing a completely incorrect location for the node. What's extra strange is that there are two separate nodes with the same incorrect location. The other node with incorrect location belongs to my friend, who I have asked about this issue. He claims he was never in the location it is reporting.

[image]

In the above screenshot, please note the node named "JAY". That would be my friend's node. What is hard to see, is my node named "B-BM" which is _beneath_ jay's node. You can somewhat see the right edge of the letter "M" beneath the "Jay" label.

Here is a view of the app, which shows both JAY and B-BM with the exact same coordinates. When in actuality, they're quite a bit closer to the RPTR node.

[image]

I will provide any local information upon request to help resolve this.

jp-bennett, 2024-04-09:
Do the nodes in question have their position precision values set to less than full precision? Adding a visualization of this to the Android map is a todo, but that's my guess as to what's happening.

djripcord, 2024-04-11:
> Do the nodes in question have their position precision values set to less than full precision?

I can look into this and get back. I am away for business but I have two nodes with me I can use to test.

closed 2024-04-28 by andrekir

swampdogmash, 2025-08-10:
I am seeing the same issue here in android. In fact, I have a node that is offline for 12hrs that is showing over the top of a friend's node. 

Position on all my nodes is set to be precise and the offline node has a GPS module. All others have fixed locations set.

jamesarich, 2025-08-10:
> I am seeing the same issue here in android. In fact, I have a node that is offline for 12hrs that is showing over the top of a friend's node.
> 
> Position on all my nodes is set to be precise and the offline node has a GPS module. All others have fixed locations set.
> 
> Interestingly I just discovered that whilst the gps module was attached, it was not locking any satellites. It was thus reporting 0 locks. Once it had locks it jumped to the correct place.
> 
> What is the mechanism for plotting a 0 locks client in the maps? Is this what explains the overlaying ("0 locks = zero position so where should I map it? Is the logic to map to the next client I'd so that it does at least get mapped?)
> 
> Andy

what version of app, firmware, phone, and phone os, any logs?