meshtastic/Meshtastic-Android #959: UI update, enhancements

lucidnx, 2024-04-09:
Hello! It seems that Meshtastic got some scale since few years ago, there are 100+ nodes in mesh especially when using MQTT and it is starting to be hard to use..
My list of top design enhancements/needed functions:
1. Move some texts and data from node list to subpage/menu made for node with all the data. Now texts are overlapping and not clear what is what.. kinda clusterf*ck
2. give user option to select shown telemetry per node on node list page
3.  move navigation to bottom so it's easy to reach or add slide from left<>right to switch between nodes, chats, settings...
4. filtering and search in node list and debug log
5. follow modern design standards maybe?

closer info for 1st and 2nd:
Node list cleaner, with just simple battery % and user selectable value (1 or 2) max per node
Node list filtering, MQTT, 0 hop, per channel.... add favourite nodes
After tapping on node, open new UI with node informations (node ID, name, etc), telemetry data, and optional buttons for triggering GPIO or so, add buttons: traceroute, ping, message, ignore, "soft" ignore (in app only, so we can ignore more nodes even if it's only chat, because radio itself can not do more than 3)
Add some graphs, history for map, telemetry data, battery levels there

I can not describe it better, hopefully it's understandable. This is how I think it will work best.

Pytlicek, 2024-04-09:
+1 for this.

lucidnx, 2024-04-13:
Oh, and some bold/highlighted chat for new messages will be nice too, +bigger channel/chat name

closed 2024-06-02 by andrekir

lucidnx, 2024-06-02:
@andrekir Oh, really? We all want more features, app up to date with iOS counterpart, modern design and you just close this with no explaination?