meshtastic/Meshtastic-Android #965: [Feature] Add ability to change channel index

djripcord, 2024-04-11:
A really convenient feature would be the ability to change channel index on the fly without the need to remove existing channels. On the android app, it seems the default behavior when adding a channel is to replace the current primary. This is problematic because it causes confusion when sharing channels with others. Especially if they're new to Meshtastic, or don't realize what happens to the channels when adding.

At the very least, if it's problematic to change channel index on the fly, we should be able to choose the channel index number when adding a channel. That would at least eliminate the primary channel replacement issue.

garthvh, 2024-04-11:
Channels don't use index other than in client apps, it is not sent over the mesh.

djripcord, 2024-04-11:
> Channels don't use index other than in client apps, it is not sent over the mesh.

This is not a request to change how data is sent over the network. It is a request to add the ability to change or swap the primary channel for a secondary. For background node discovery via automatic telemetry reports, LongFast should be the primary channel. But if a user wishes to use a different channel for telemetry, it should be easier to do so. Just try adding a new channel to a node. You will see it puts LongFast to secondary and the new channel is set to primary.

I think we should be able to choose whether or not to set a new channel as a primary upon it's creation or import.

andrekir, 2024-06-30:
long-press drag and drop was added to the channel editor in 2.3.14

closed 2024-06-30 by andrekir