meshtastic/Meshtastic-Android #969: [Feature request]: Compatibility with Android Auto

mgoljak, 2024-04-15:
Hello Meshtastic Team,

The main idea is to allow the Meshtastic Android app to be used directly from the Android Auto interface. This would provide users the ability to access the app's features directly from their car's infotainment system, making it more convenient and safer to use while driving.

Thank you for your time and consideration.

Best regards,
Mario

infered5, 2024-10-13:
This will require some code changes to meet the strict guidelines put into place by Google for Android Auto (app with a paired phone) and Android Automotive (Android running on a head unit itself).

https://developer.android.com/docs/quality-guidelines/car-app-quality
https://developer.android.com/training/cars/distribute

This will push the minimum Android level requirements up to Android 9. 

Some additional notes based on my knowledge of Android Auto:

* Message replies using a full keyboard MUST only be possible when parked. (This is possible, but needs code.)
* Google must approve all app updates with a further in-depth review process than standard Play Store apps.
* Buttons that do canned responses _should_ be okay to use without parking. (Someone more versed in AA please verify this).
* Freeform responses must have the ability to use Speech to Text using Google Assistant.

https://developers.google.com/cars/design/android-auto/apps/images/conversationalflow.png

This will take a bit of work.

Jos1990, 2024-10-16:
This would be very helpful. I had the exact same question.

Kentajoh, 2025-04-09:
Perhaps only a map view for tracking nodes?

closed 2025-05-24 by github-actions[bot]