meshtastic/Meshtastic-Android #981: Show colour on rssi and snr of the node its received through

djcasper1975, 2024-04-19:
When looking at rssi and snr it can be confusing and look like signal is really good when infact your actually receiving them through other nodes. so if node 0 is me and i see node 3 through node 1 then show rssi and snr the colour of node 1. this does change as nodes come though other nodes so this would change when refreshed if its a new node i,e now 0 receives nodes though node 3 through node 4 etc. shouldnt be too hard to do and would give users more of an understanding as to how this is all working when they see that node all nodes refresh eachother from the same nopde could be direct (white) or through another node (node colour)

GUVWAF, 2024-04-20:
There is an open PR #966 to only show the RSSI and SNR for a directly connected node and otherwise the amount of "hops away".

Except for traceroute, we don't know who relayed a packet for us, so we can't update their signal quality based on this.

djcasper1975, 2024-04-20:
> There is an open PR #966 to only show the RSSI and SNR for a directly connected node and otherwise the amount of "hops away".
> 
> Except for traceroute, we don't know who relayed a packet for us, so we can't update their signal quality based on this.

well im not a computer but i can clearly see the node im receiving through as it has almost the same snr and rssi on all the nodes going through it. give me somewhere to send video and ill point it out i have 1 node shows all nodes in its town. and all them nodes have almost the same rssi and snr as the one its coming through ............. so why cant it show the colour of that node on the text or rssi and snr saves having to make room for more test for hops. ive looked at the code and im no android expert but tell me where that text can be altered in your code and ill find a way to do it for myself then show you. where do i have access in the code to where i can change rssi and snr colour.

djcasper1975, 2024-04-20:
> There is an open PR #966 to only show the RSSI and SNR for a directly connected node and otherwise the amount of "hops away".
> 
> Except for traceroute, we don't know who relayed a packet for us, so we can't update their signal quality based on this.

    /**
     * ```
     *
     * Returns the Signal-to-noise ratio (SNR) of the last received message,
     * as measured by the receiver. Return SNR of the last received message in dB
     * ```
     *
     * `float snr = 4;`
     */
    public var snr: kotlin.Float
      @JvmName("getSnr")
      get() = _builder.getSnr()
      @JvmName("setSnr")
      set(value) {
        _builder.setSnr(value)
      }
if you see the signal of message through reciever cant the recivers colour be shown on rssi and snr..... if its not you... surly it can be done.

GUVWAF, 2024-04-20:
> if you see the signal of message through reciever cant the recivers colour be shown on rssi and snr..... if its not you... surly it can be done.

The RSSI and SNR change per packet, even for the same node, so how similar should they be?

How do you know which node it was if you have multiple nodes with roughly the same RSSI and SNR?

djcasper1975, 2024-04-20:
> The RSSI and SNR change per packet, even for the same node, so how similar should they be?
>
> How do you know which node it was if you have multiple nodes with roughly the same RSSI and SNR?

It must know somehow surly the last node it was recived from otherwise it wouldnt show the rssi and snr of that node its reciced through. So when i look at node list i see 4 all same rssi and snr ans when i do trace rt its the same node im receiving them through.  So if that was the colour of that node that would be great. This system  is so blind. Also many are leaving because they cant message anybody unless on long slow which bobodys on so...... but back to the point it looks way nicernin the app with colours and helps know where the last packet came through from when looking at the nodes.

prokrypt, 2024-04-20:
The RSSI/SNR values are for individual received packets, not nodes. The only thing known is the node that created the original packet (in the "from" field of the packet). Nodes that rebroadcast the packet only decrement the hop limit counter and nothing else in the packet is changed.

closed 2024-04-21 by andrekir