meshtastic/Meshtastic-Android #982: [BUG] 2.3.4 When connected to a device via ip, and switching to bluetooth, you can't switch back to the IP agian

Marx1, 2024-04-20:
Description: If you're connected via IP to a device, then switch to another via Bluetooth, and try and switch back, the IP address isn't saved; and you can't enter it in the text field/hit DONE to connect. You have to empty the text box, set to none then set the ip and re-connect.

See attached video demonstration

Expected behavior: The IP remains in the list and you can freely switch between them without having to delete it and re-add it every connection.

Device: Samsung S22 Ultra (Unlocked)
OS: Android 14, security patch March 1 2024

https://github.com/meshtastic/Meshtastic-Android/assets/4861651/f204d5d8-68af-46fb-b62d-3a3888a6f205

prokrypt, 2024-04-20:
Duplicate of https://github.com/meshtastic/Meshtastic-Android/issues/691

Anyways, there is no need to click None. My preferred method is to input an additional number to the IP and then backspace to re-enable the item selection.

Marx1, 2024-04-21:
While some of this bug overlaps with that FR, this issue is not really a FR, it's a bug. you can't interact with the IP - ie you can't change the ip with out clearing it. If you're re-using the same ip - yes it works, you can't re-connect to an ip without editing it, it shows the ip in the list, but doesn't save it. 

All of these are bugs, as it's something that exists, but doesn't work.  if IP was missing, then it would be a feature request, or if it didn't show up in the list; as if it saved it, and you had to enter an IP each time; requesting it to save it would be a feature request.

prokrypt, 2024-04-21:
I've decided to fix this little quirk though I don't know if I did it right...

closed 2024-04-21 by andrekir