meshtastic/Meshtastic-Android #987: User-friendly region selection

AndreiRegiani, 2024-04-21:
The initial setup can be more friendly by displaying country names in the region selection:

**ACTUAL**

[image]

---

**EXPECTED**
Code + country name, as shown in the official documentation:

[image]

closed 2024-10-09 by andrekir

AndreiRegiani, 2024-10-09:
>ANZ(RegionCode.ANZ, "Australia / New Zealand", 915.0f, 928.0f),

Could we add `Brazil` too? it's the same frequency, and has a big community.

https://meshtastic.org/docs/configuration/region-by-country/

andrekir, 2024-10-09:
Brazil uses ANZ similar to how Canada and other countries use US for their frequency plans. The description reflects the primary regulatory region, not an exhaustive list of countries using it.

commit 9cce7855 (2024-10-09): feat: add `RegionCode` descriptions to region selection

closes #987