meshtastic/Meshtastic-Android #994: [Feature Request]: Display INA3221 measurement values on app

geekfuns, 2024-04-24:
### Platform

ESP32

### Description

Hello, I have connected an INA3221 sensor to my ESP32 board and it is working well. Its measurement values can be seen in the debug information from the serial port and also the screen, but they cannot be seen in the app. The multi-channel measurement function of INA3221 is very useful for solar powered  stations, as it can read the voltage and current of the solar panel, as well as the working current of the system. May I ask if this feature can be added?
May the feature is already supported, can you tell me how to configure it?
Thanks a lot!

geekfuns, 2024-04-25:
> > The mashtastic app. Sorry, I posted this issue in the wrong place. Please help me move it over. Thank you
> 
> Are you referring to the iOS app or Android one?

I am using an Android application, and after replacing INA3221 with INA219, I can read the measurement values on the app.

valzzu, 2024-05-04:
Id like this too. 
Its connected to a wisblock with no screen 😅

closed 2024-10-01 by andrekir

xia0, 2025-11-25:
> > > The mashtastic app. Sorry, I posted this issue in the wrong place. Please help me move it over. Thank you
> > 
> > 
> > Are you referring to the iOS app or Android one?
> 
> I am using an Android application, and after replacing INA3221 with INA219, I can read the measurement values on the app.

INA219 works for me but so does INA3221 after changing the address to 0x42 (details here https://github.com/meshtastic/firmware/issues/6935#issuecomment-2925726188)

commit cca06784 (2024-10-01): feat: add `PowerMetrics` data to nodes telemetry

closes #994