meshtastic/Meshtastic-Android #998: [Feature Request] Add more columns to rangetest.csv (in particular RSSI)

barryhunter, 2024-04-26:
I think it would be really useful to add some extra columns to rangetest.csv output, to make it more useful. 

In particular `rx rssi` and the `to` node number. 

It appears the SNR is already just included by decoding to protocol buffer, so should have rxRssi, similar to having rxSnr
https://github.com/meshtastic/Meshtastic-Android/blob/b409c17fe8276026586a7dcf932b45bc4479cee7/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L1021

... and the destination node (proto.to) could be very useful too, particular to find if the message was intended for your node or not (and distinish broadcast packets) - as can receive packets for others (if have the decruption key for the channel!) 

Seems could also add 'position age' (sender and rx!), eg in seconds - although can in theory deduce it yourself for sender at least (as see the POSITION_APP messages!) - the 'rx' position packets are filtered (no snr) - so maybe only need `rx position age`

I could modify the java code, to add these columns (to the end!), but to be honest, not sure if I am upto trying to recompile the app to test it. Never built an android app. 

As an aside, wonder if would make sense to dump the payload for <TRACEROUTE_APP>, less sure about <NEIGHBORINFO_APP>
... probably <TELEMETRY_APP> too, but have less interest in that myself. But would make rangetest.csv a way to get the telemetry log, not sure if any other way to extract the historic data. 

... would also be nice to extract node names from <NODEINFO_APP> - while the 'sender name' column is populated from from the *current* node database, old packets will be missing (nodes have been forgotten, but still have messages in database 'log' table)

andrekir, 2024-04-28:
we just mirror firmware code for the columns: https://github.com/meshtastic/firmware/blob/master/src/modules/RangeTestModule.cpp#L239

adding RSSI has been discussed before, see: https://github.com/meshtastic/firmware/issues/1215 and https://github.com/meshtastic/firmware/issues/2372

closed 2024-04-28 by andrekir

jibinbabuamakkattu, 2025-02-27:
i mirrored the same and added a coloumn named temperature.
fileToWrite.println(
                "time,from,sender name,sender lat,sender long,rx lat,rx long,rx elevation,rx snr,distance,hop limit,temperature,payload");
but the problem is that, still when i download the csv, it doesnt contain the new header.
Know why its happening?

jibinbabuamakkattu, 2025-02-27:
And another doubt i had is that, does it create a new header if the old rangetst.csv exists from the previous sessions? like ive been testing it out, and even after i flash a new firmware and download the csv, the data from the old csv exists in it. could it be the reason that the new header isnt being written? if yes , how do i rectify it?