meshtastic/Meshtastic-Android #1001: [Feature Request]: Implement Android Notification Channels

David-Woodward, 2024-04-27:
Implement Android Notification Channels for each LoRa channel configured on the device so that we can mute notifications from specific channels.

https://developer.android.com/develop/ui/views/notifications/channels

For example, we could mute LongFast notifications and allow other private channel notifications to come through.

andrekir, 2024-04-28:
this is not the way to go. standard messaging app behavior is long-pressing a contact and selecting a mute icon. there's an open feature request for this: #511

closed 2024-04-28 by andrekir

David-Woodward, 2024-04-28:
I disagree.  The default Google Messages (SMS texting) app uses notification channels for SMS groups exactly as I've described.

If we have both a "Tech Talk" channel and my "Politics" channel, it makes more sense to mute the Politics channel when we need a break from political debates than to mute individual nodes in the channel.  That not only takes a lot of time, but it also would mute both the undesirable political banter as well as the tech talk if the nodes are in both channels.