meshtastic/Meshtastic-Android #1002: Network connection not working.

lemonade1947, 2024-04-27:
Hello. I am using 
version 2.3.4 of the app
version 2.3.6 board firmware. 

Have tried alpha 2.3.5 version of app, and this issue seems to exist there too. 
Have tried complete reset of board, no change. 

Connection over network was previously working fine, but sometime last night it stopped working. 

I believe this is a problem with the app, not the firmware, because I was able to connect to the boar using the web client with no issues. 

I've done quite a bit of digging here, although I've little idea what I'm doing. 

[image]

Sorry that this is a screenshot. I wish I'd pulled the text from it but unfortunately I didn't. It seems, obviously, that something to do with SSL is failing. I really am not sure what to make of this. As I've said, this was certainly working just a day ago. 

All I could think of was that the time was set wrong on either the phone or the board, but it appears they're correct in both cases. 

Any help would be appreciated.

andrekir, 2024-04-28:
there are no changes to the TCP/IP interface in the last 6 months, and SSL is not used (HTTP only).

andrekir, 2024-04-28:
flashed 2.3.7 firmware on a T-Beam and it's been running for an hour without any issues.

make sure you're not trying to connect more than one client at a time, that's not supported in the firmware.

closed 2024-05-05 by andrekir