meshtastic/Meshtastic-Android #1005: [Feature Request] ability to apply nicknames to nodes

faradayfedora, 2024-04-29:
Think it would be nice if you could apply your own name to node, use case is putting know names since alot of nodes are aliases.

prokrypt, 2024-04-29:
What may be easier to implement (and more useful in my opinion) is a searchable notes field for each node...

closed 2025-05-24 by github-actions[bot]

arugulax, 2026-10-21:
Just adding my two cents, this functionality would be tremendously helpful and seems like it would be trivial to implement (I haven't looked at the codebase, granted). It doesn't have to go out over the network, just a field saved on your local device as an alternate node definition. Or, you could have the "notes" field pop up with the first click on the node on the map. Less ideal, but I'll take what you offer! Thanks for your reconsideration of this feature.