meshtastic/Meshtastic-Android #1011: [Feature Request] Favorite Nodes in Android

dustingrady, 2024-05-01:
I believe in iOS there is the ability to "favorite" a node, so that it does not fall off of your NodeDB. It would be nice to have something similar in Android :)

mudcamper, 2024-06-29:
Agree good idea. There is so much activity in the Bay Area that all my local nodes drop off overnight while 200+ nodes far away clutter up my node dB.

mudcamper, 2024-07-03:
andrekir on the iPhone app you press and hold in the node list and you get a menu that includes Add Favorite. On Android it could go in the same menu in the node list you get from tapping on a node name.

esev, 2024-11-05:
FWIW without this, it's hard to use the `KNOWN_ONLY` Rebroadcast Mode. The devices I'd like to keep in my DB end up getting overwritten after it fills.

Kealper, 2024-11-25:
Also interested in this feature now that my area is starting to get enough nodes and infrastructure that I'm regularly having my nearby nodes cycle out.

PR #1520 (2025-02-16): feat: Initial implementation of adding nodes to favorites
Initial implementation of support for favorite nodes. Adds support for and closes #1011

I've tested this in the simulator while it was connected to a real node and it seems to work as you'd expect, but it'll need a second pair of eyes on it because I'm not familiar with Kotlin and this is my first time working with it.
closed 2025-02-16 by jamesarich