meshtastic/Meshtastic-Android #1023: devices don't get to know each other

cracky22, 2024-05-04:
I currently have 2 repeaters in use and my own handheld device. There are additional repeaters and other handheld devices in the mesh. But my handset only knows one of 2 repeaters. what can I do so that my handset knows the 2nd repeater? The messages still go (traceroute: unknown device) via the repeater (presumably)

GUVWAF, 2024-05-04:
So it is in `REPEATER` [role](https://meshtastic.org/docs/configuration/radio/device/#roles)? Repeaters won't show up in the node list as the documentation says (unless if was another role before and your device already heard it). 
Set it in `CLIENT` role if you want them in the node list. Clients will also repeat messages.

cracky22, 2024-05-04:
oh interesting, thanks for the info

closed 2024-05-04 by cracky22