meshtastic/Meshtastic-Android #1029: Issues with Use fixed position

Bryang0syr, 2024-05-10:
Using Firmware 2.3.6 ad Android 2.3.7 though suspect occuring in earlier releases. If I set my position and height manually and have 'Use fixed position' turned on, if I connect my Android Galaxy A13 with app permissions for Meshtastic on then the manually entered figures are overridden and the phones GPS data is used. If I turn off Meshtastic's app permissions of then the node reverts to the manually entered figures. When 'Use fixed position' is in operation although the height ASL shows correctly when manually entered it falls to zero at the next position broadcast. The correct manually entered height is displayed if the node is rebooted for any reason but again falls to zero at the next position transmission.

caveman99, 2024-05-10:
for settings these values the 'fixed position flag' must be off. turn off - set values - turn back on

Bryang0syr, 2024-05-10:
Sorry I don't understand If I turn Use fixed position off where do I enter the values as the boxes for fixed positions are not visible ?

andrekir, 2024-06-08:
this should no longer be an issue

closed 2024-06-08 by andrekir