meshtastic/Meshtastic-Android #1034: Export configuration not working + weird behavior on Android 9

Overlord23, 2024-05-12:
Hi! 
I'm running the app on Android 9 connected to my ESP32 based lora board.
When I hit _Export configuration_ the exported file is corrupt, opening with an hex editor seems to contain only the channel URL, NTP configuration along with some random data, the size is around 270 B; a proper configuration file downloaded from CLI is around 1,5 KB
if I hit _Export configuration_ again a popup message with _Delivery confirmed_ appears on screen, thereafter the board reboots.
On the serial output seems that the app sent a configuration to the radio causing the restart above.
This time there isn't any file saved on my phone.

Next I tried from an android 14 and the exported file is still corrupt, on this android version i can save export the configuration every time without getting the _Delivery confirmed_ bug.

garthvh, 2024-05-13:
The cli file is yaml and the android app outputs a binary file so they are not going to match in size

Overlord23, 2024-05-14:
I've restored a configuration made from app on Android 14 successfully. @garthvh  you are right, the configuration file is not corrupted.
On Android 9 instead every time I try to export a configuration, the exported cfg file is 0B size.
Rebooting the phone doesn't make difference, also exporting configuration from a lora board connected either via bluetooth and http, the issue is still the same

closed 2024-06-08 by andrekir