meshtastic/Meshtastic-Android #1037: Add 'Update Firmware' option

chrisdebian, 2024-05-14:
Related to #1031 

When a device is connected, check the firmware version, and give the user the option to download and flash firmware. I suspect many users will not have updated the firmware, since original purchase/ flash.

Thanks,

Chris
2E0FRU

madeofstown, 2024-06-02:
Not sure if this could work over Bluetooth or WiFi as I don't believe Meshtastic has the ability for OTA updates like that. Also, lots of devices need to be manually placed in a flashing mode before they can have their firmware updated.

prokrypt, 2024-06-02:
I believe earlier versions of the android app were able to update esp32 devices over Bluetooth. I remember doing it for a t-beam...

Talie5in, 2024-06-04:
> Not sure if this could work over Bluetooth or WiFi as I don't believe Meshtastic has the ability for OTA updates like that. Also, lots of devices need to be manually placed in a flashing mode before they can have their firmware updated.

Biggest issue is space available on most devices (4MB Flash) - is not enough to fit the running Firmware (2.2MB on most devices) and a second copy in the ota_2 partition to swap over to after sending up the new image.

At least thats how I understand it, i've been messing around in this space in a fork and that's basically what i've run into as been the real show stopper here...

madeofstown, 2024-06-06:
@Talie5in I see you have it working now in your fork. When are you doing a pull request?

Talie5in, 2024-08-01:
> @Talie5in I see you have it working now in your fork. When are you doing a pull request?

@madeofstown Sorry, not sure how I missed this comment!

It does, however, as per my comment above, is not suitable for 4MB flash devices, it's a space limitation thing, and I only worked on the WiFi portion (over the webserver) and I did have to recraft the partition layout and make minimal builds (Eg, no screen, no modules, etc)

Short of, making the firmware much smaller or devices starting to come out with 8MB flash as a standard..

Alternatively, having the BLE_OTA firmware able to bootup and support WiFi/BLE from existing meshtastic config and a small webserver (for over WiFi) - this is the key, been booted into ota_1 so we can write to ota_0 partition :)

jaredcat, 2024-08-05:
At the very least, a warning letting users know there's a new update would be helpful!  
Then down the line can open up OTA to those devices that support it.

cracky22, 2024-10-27:
> Not sure if this could work over Bluetooth or WiFi as I don't believe Meshtastic has the ability for OTA updates like that. Also, lots of devices need to be manually placed in a flashing mode before they can have their firmware updated.

only on nrf52 devices

kquinsland, 2024-12-31:
> Also, lots of devices need to be manually placed in a flashing mode before they can have their firmware updated.

When the user is manually doing so, yes. But app code can - if it wants - write anything to any where in flash and then initiate a reboot. Before rebooting, it's possible to indicate which partition should be booted into if needed.
While not ideal, it's not unheard of to basically do just that: upload a shim, write that to the smallest partition and then boot into it. Shim does nothing but wait for "regular" partition to be uploaded. When the last chunk has been written to the main partition, reboot back into app land and the OTA is done :).

closed 2025-05-24 by github-actions[bot]