meshtastic/Meshtastic-Android #1042: [Feature Request]: Clear display of IQA.

memo-567, 2024-05-15:
It would be nice if there was also an understandable display of the IQA for the Android app.
Currently, the value is displayed in `MΩ` only.

This is already available for the iPhone app: https://github.com/meshtastic/Meshtastic-Apple/commit/52dd086e09b7b7e6d1e5e693e90e6db38fe98501

At the moment it looks like:

[image]

grenius, 2024-05-25:
That's power telemetry isn't it? Though i agree a weather telemetry page like iOS would be nice.

memo-567, 2024-05-25:
> That's power telemetry isn't it? 

This value the BME680 sends as IAQ (Indoor Air Quality) https://en.wikipedia.org/wiki/Indoor_air_quality

grenius, 2024-05-25:
bme680 only has pressure/temp/humidity.

memo-567, 2024-05-25:
Nope: https://www.bosch-sensortec.com/products/environmental-sensors/gas-sensors/bme680/

grenius, 2024-05-25:
Oh my bad. Must have been thinking of the 280. Either way i'm onboard with what you're asking.

grenius, 2024-05-25:
FYI:

MΩ
A unit of resistance in electricity that is equal to one million ohms, also known as a meg

memo-567, 2024-05-25:
Thanks for the tip, but I'm very familiar with the basics of electrical engineering 😉

andrekir, 2024-06-30:
IAQ values were added in 2.3.12

closed 2024-06-30 by andrekir