meshtastic/Meshtastic-Android #1044: When I clear the node DB my fixed position values are reset to 0

dougjeff, 2024-05-17:
For my home node I use fixed position instead of having a GPS that draws power.
When I set values for the lat, long & alt they appear correctly.
When I do Radio Configuration>Node DB reset and send it my lat, long & alt are reset to 0.

Running Heltec V3 fw 2.3.3
Android app v 2.3.7

I would not expect fixed positions to get reset like this. Or is this the intended behaviour?

Thanks
Doug

prokrypt, 2024-05-18:
It is the expected behavior for those of us in the know, but I guess it can be considered a bug because everyone else doesn't expect it...
Perhaps a change can be made in the firmware to keep the position when resetting nodedb.

caveman99, 2024-05-18:
PLease update your device firmware. There was a recent change to not clear node 0 (my own record) when resetting nodedb.

caveman99, 2024-05-18:
Jzst to amend: it keeps the own node in DB but explicitly resets its position data. So disregard my previous comment, this is intended behaviour.

caveman99, 2024-05-18:
https://github.com/meshtastic/firmware/pull/3451

thebentern, 2024-05-18:
Yes this is working as designed.

closed 2024-05-18 by thebentern