meshtastic/Meshtastic-Android #1046: Bluetooth not reconnecting when coming back within range

Bryang0syr, 2024-05-19:
Can't be certain but a few days ago when returning instead of reconnecting I now have to re-pair the phone and node each time entering the fixed pin. Not sure when Android 2.3.8 was pushed out but started to notice the problem a few days ago whereas before the reconnection was automatic, running 2.3.6 firmware on Heltec V3

todd-herbert, 2024-05-28:
Are you still using the same firmware on your node? The only thing that changed was the app version?

pomazanbohdan, 2024-05-28:
Personally, I just started using it Application version 2.3.10

andrekir, 2024-06-08:
losing pairing is not an app issue

closed 2024-06-08 by andrekir

todd-herbert, 2024-06-08:
Just to follow up on https://github.com/meshtastic/Meshtastic-Android/issues/1057#issuecomment-2134553637, I haven't noticed it again since then. Probably something had just gone weird on my end for a bit.