meshtastic/Meshtastic-Android #1062: [Feature Request]:  Android client share feature.

clavo2, 2024-05-31:
### Platform

ESP32

### Description

Would it be possible to add a android 'share' feature to the app so it shows up as an app that accepts shared data?  I realize droid can share more than text (pictures, etc) between apps, but curious if just text based shares could be sent to the android app to save from cutting text from one app, changing app, and pasting.  

Pardon my ignorance, but this may allow workflow apps like Tasker to automatically route things to the Meshtastic app thus automating certain things (e.g. incoming SMS texts could be pushed to the Meshtastic app and automatically be send to a predefined channel.  

I can see this as being useful in many ways.  Like maybe in the future pushing emails (might require a separate app catered to this).   e.g. email to: mesht@aol.com - filter sees "MT" in the subject and Tasker or app native pushes that to the Meshtastic app and out it goes. 

Just an idea that may be easy, or really hard to do. 

I know this can be done with CLI, but we all live on phones and not everyone knows how to cat herd Termux for this kind of stuff. 

Thanks for the consideration.

jamesarich, 2024-06-28:
https://developer.android.com/training/sharing/receive
Docs here for setting up share intent. Would we want to try and provide Sharing Targets to recent channels and node dms?

remi-espie, 2024-11-06:
Hello, I'm interested in working on this issue. Could you please assign it to me ?

jamesarich, 2024-11-06:
> Hello, I'm interested in working on this issue. Could you please assign it to me ?

I don't think anyone is currently working on this - feel free to start a draft pull request outlining what you're trying to accomplish.

closed 2024-11-30 by andrekir