meshtastic/Meshtastic-Android #1086: Android/Apple Client Disparity, Cannot Share PSKs of 192bits

AddisonTustin, 2024-06-11:
## Summary
It does not seem possible to share PSKs between Apple/Android clients when the PSK is created/generated by the Apple client with a key size of 192 bits.

## Steps to reproduce
- In Apple client, create a new channel with an auto-generated key of 192bits.
- Share that PSK to an Android client/user, and have them attempt to create a channel by pasting the PSK. 

There's no warnings or error messages for why this is an invalid channel key for Android, while it's supported on the Apple client.

## Apple Image
[image]
## Android Image
[image]

AddisonTustin, 2024-06-11:
Trying to spin-up on some of the Android package, but is there any reason we shouldn't be able to accept an additional [key size here](https://github.com/meshtastic/Meshtastic-Android/blob/5b3c78316b8accc789471f7ebdcd36e295c76682/app/src/main/java/com/geeksville/mesh/ui/components/config/EditChannelDialog.kt#L102) of 24 bytes?

andrekir, 2024-06-13:
firmware doesn't support 192 bit keys

https://github.com/meshtastic/protobufs/blob/master/meshtastic/channel.proto#L36
https://github.com/meshtastic/firmware/blob/master/src/mesh/CryptoEngine.h#L38

AddisonTustin, 2024-06-13:
I'd take a look at the [comment @ianmcorvidae](https://github.com/meshtastic/Meshtastic-Android/pull/1087#issuecomment-2162340971) added to #1087. 

The firmware pads the key to be 256-bits and the [Meshtastic-Apple client](https://github.com/meshtastic/Meshtastic-Apple/blob/a4e26de6e5423ddeb70d40388d80690006f21bb1/Meshtastic/Views/Settings/Channels/ChannelForm.swift#L60) 'accepts' a 192-bit key.

closed 2024-06-30 by andrekir