meshtastic/Meshtastic-Android #1095: Feature Request:  Resistance or IQA display by BME680

memo-567, 2024-06-13:
After updating to version Meshtastic Android 2.3.12 alpha, both the resistance value and the IAQ are displayed.

If both can be displayed, can this possibly be limited to the IAQ?

garthvh, 2024-06-13:
This is a good point resistance means nothing to anyone anyways, only the IAQ is useful to humans

Donnie650, 2024-06-13:
Thats not a true statement, some of us enjoy having All the data that a sensor can output. Maybe allow a switch to turn it on? I'm a numbers person and enjoy all the data.  Please fine a way for both to enjoy output.

garthvh, 2024-06-13:
> Thats not a true statement, some of us enjoy having All the data that a sensor can output. Maybe allow a switch to turn it on? I'm a numbers person and enjoy all the data. Please fine a way for both to enjoy output.

It is, the value needs additional processing to be meaningful.

b8b8, 2024-06-13:
Can the text be colour coded too, to the same as Garth's iOS app?

garthvh, 2024-06-14:
> Can the text be colour coded too, to the same as Garth's iOS app? edit: or maybe a little colour coded dot next to it

Most of this is not in the iOS app yet, but here are some controls using the IAQ scale.

https://github.com/meshtastic/design/issues/18

garthvh, 2024-06-19:
> What I mean is its not true that people have no use for it. I do, and I'm sure others would like all the data a sensor can provide. Just an opinion.

You can't interpret the value without additional processing, like raw gps coordinates no one is doing this calculation in their head and it does not need to be displayed to humans.

Donnie650, 2024-06-19:
Will it still be part of the data sent out and just ignored by the app? I do proccess all the data I get and would like to be able to continue to do so with the python cli, rewote meshwatch to do all this. Hope it will remain available in the firmware. If not thats ok too. Thanks for your replys.
[image]

andrekir, 2024-06-30:
this can be personal preference and if data has no value it really shouldn't be sent over LoRa

closed 2024-06-30 by andrekir